arduino/arduino-fwuploader #4: Wifi101 Bricked — Firmware Updater loop?

wrongtarget, 2016-03-02:
This is all I get when attempting to upgrade the firmware of the Wifi 101 on my official Arduino Uno:

2016/03/02 14:56:10 Connecting to programmer
2016/03/02 14:56:12 Synch with programmer
2016/03/02 14:56:12 Reading max payload size
2016/03/02 14:56:12 Flashing firmware from '/Wifi101_FirmwareUpdater/firmware/19.4.4/m2m_aio_2b0.bin'
2016/03/02 14:56:12 Error during FlashErase()

Now none of the examples, including Check101FirmwareVersion work.  Could had something gone wrong during the upgrading process? Anything I can do to start anew?

sandeepmistry, 2016-03-03:
@wrongtarget have you tried the upgrade procedure again after the failure?

wrongtarget, 2016-03-03:
Yep, same error unfortunately.

sandeepmistry, 2016-03-03:
@wrongtarget I just tried this out on my Uno + WiFi101 shield, no errors. I was using version 0.8 of the WiFi101 library.

What version of the library are you using?

wrongtarget, 2016-03-03:
Same version. Latest one — I got the WiFi101 yesterday so I made sure to have the latest library by then.

wrongtarget, 2016-03-03:
My bad — the shield was not fit properly into the Arduino. I was being too careful when pressing down.
It's now properly detected.

closed 2016-03-03 by wrongtarget

sandeepmistry, 2016-03-03:
No worries, thanks for following up on the cause of the issue :)