arduino/arduino-fwuploader #5: Firmware Updater stuck on "Synch with programmer"

PKGeorgiev, 2016-05-03:
I'm trying to add new certificate to my mlk1000. My steps were:
1) I've uploaded FirmwareUpdater sketch to the MCU with no errors
2) I ran winc1500-uploader.exe -address=*****.com:443 -port=COM4
3) The result is:

2016/05/03 17:45:31 Connecting to programmer
2016/05/03 17:45:33 Synch with programmer

The same happens with winc1500-uploader-gui. It stucks on "Synchronizing with the programmer".

My configuration:
1) IDE selection
  * Board: Arduino/Genuino MKR1000
  * Port: COM4

2) OS Windows 10

PKGeorgiev, 2016-05-16:
Interestingly, after soldering the two racks (for testing gpio pins) - I was able to upload custom certificates. 
May be the problem was due to some "noise" or something. If somebody encounters the same problem - just solder the racks and try again.

p.s. I haven't tried lower baud rates. May be 115200 bps is too high and make the connection unstable?

sandeepmistry, 2016-05-27:
@PKGeorgiev 

> If somebody encounters the same problem - just solder the racks and try again.

Could you please expand on what this mean?

> p.s. I haven't tried lower baud rates. May be 115200 bps is too high and make the connection unstable?

Since the MKR1000 uses native USB for the Serial port, the baud rate is not actually used.

PKGeorgiev, 2016-05-29:
@sandeepmistry after soldering the two male strip headers to the MCU I was able to upload certificates to mkr1000 without problems. 
My theory is that sometimes there may be some interference at 115200 bps. I cannot prove it, because I have only one MKR1000 at the moment.
I've seen other people experiencing the same problem.

sandeepmistry, 2016-05-30:
@PKGeorgiev there's most likely something software related. The h/w pins are not used for the USB data transfer for firmware upload.

> I've seen other people experiencing the same problem.

Where have these been reported?

I know there is an open issue with OS X and firmware updates over native USB ports (which something like https://github.com/arduino/ArduinoCore-samd/pull/116 resolves).

PKGeorgiev, 2016-05-31:
@sandeepmistry , my OS is Windows 10. I've found some comments like http://mohanp.com/mkr1000-azure-iot-hub-how-to/#div-comment-37

It is fact that I wasn't able to upload certificates without first soldering the two male strip headers to the MCU. That's why I suggested more verbose output in the CLI.