arduino/arduino-fwuploader #12: segmentation violation

arwie, 2020-09-25:
I just tried to update the firmware of my nano iot 33 board. The updated failed and left my board unusable (no /dev/ttyACM0 any more). How can I restore my board?

./updater -flasher firmwares/NINA/FirmwareUpdater.mkrwifi1010.ino.bin -firmware firmwares/NINA/1.4.1/NINA_W102.bin -port /dev/ttyACM0 -programmer /home/client/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac 

2020/09/25 17:03:14 Flashing firmware uploader nina
2020/09/25 17:03:14 Flashing firmwares/NINA/FirmwareUpdater.mkrwifi1010.ino.bin
2020/09/25 17:03:14 Restarting in bootloader mode
2020/09/25 17:03:14 Get port list before reset
2020/09/25 17:03:14 Touching port /dev/ttyACM0 at 1200bps
2020/09/25 17:03:14 Set DTR off
2020/09/25 17:03:25 Wait for the port to reappear
2020/09/25 17:03:27 Atmel SMART device 0x10010005 found
Unlock all regions

Read 253952 bytes from flash
[==============================] 100% (3968/3968 pages)
done in 1.254 seconds

2020/09/25 17:03:27 Original sketch saved at /tmp/wifiFlasher_dump183132501/dump.bin
2020/09/25 17:03:29 Atmel SMART device 0x10010005 found
Erase flash
done in 0.880 seconds

Write 13688 bytes to flash (214 pages)
[==============================] 100% (214/214 pages)
done in 0.085 seconds
CPU reset.

2020/09/25 17:03:39 Connecting to programmer
2020/09/25 17:03:41 Sync with programmer
panic: runtime error: invalid memory address or nil pointer dereference
        panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x18 pc=0x61d1e9]

goroutine 1 [running]:
github.com/arduino-libraries/FirmwareUpdater/modules/nina.(*Flasher).Close(0xc0002da440, 0x0, 0x0)
        /home/martino/FirmwareUpdater/src/github.com/arduino-libraries/FirmwareUpdater/modules/nina/flasher.go:76 +0x29
panic(0x64ca00, 0x8031d0)
        /usr/lib/go/src/runtime/panic.go:969 +0x175
encoding/binary.Write(0x0, 0x0, 0x6d3100, 0x83e0b0, 0x6411e0, 0x7f5e88, 0x29, 0x14)
        /usr/lib/go/src/encoding/binary/binary.go:375 +0x149
github.com/arduino-libraries/FirmwareUpdater/modules/nina.(*Flasher).sendCommand(0xc0002da440, 0x1122334400068099, 0x55667788, 0x0, 0x0, 0x0, 0x4f120c, 0xc000068050)
        /home/martino/FirmwareUpdater/src/github.com/arduino-libraries/FirmwareUpdater/modules/nina/flasher.go:172 +0x98
github.com/arduino-libraries/FirmwareUpdater/modules/nina.(*Flasher).Hello(0xc0002da440, 0x1, 0x1)
        /home/martino/FirmwareUpdater/src/github.com/arduino-libraries/FirmwareUpdater/modules/nina/flasher.go:48 +0x54
github.com/arduino-libraries/FirmwareUpdater/modules/nina.Run(0x7ffc7ef483a5, 0xc, 0x0, 0x0, 0x0, 0x0, 0x0, 0x7ffc7ef4837c, 0x22, 0x7ffc7ef4833f, ...)
        /home/martino/FirmwareUpdater/src/github.com/arduino-libraries/FirmwareUpdater/modules/nina/main.go:73 +0x2f5
main.main()
        /home/martino/FirmwareUpdater/src/github.com/arduino-libraries/FirmwareUpdater/cli/main.go:47 +0x218

aentinger, 2020-09-30:
@facchinm do you've got any idea about what happened here?

facchinm, 2020-09-30:
@aentinger it should have been fixed by @cmaglie patch https://github.com/arduino/FirmwareUpdater/pull/11 , which refactors the serial port handling.
We should recompile and release the updated binaries

arwie, 2020-11-10:
Just tried with V0.1.2 and it works! Thank you!

closed 2020-11-10 by arwie