arduino/arduino-fwuploader #30: RPI flashing not working

MattyBoy4444, 2021-05-12:
I am using a RPI3.  When I run the command with the correct serial device name it just says 
'no device found'.  Has anyone has the same problem?

`./FirmwareUploader -flasher ~/firmwareuploader/firmwares/NINA/FirmwareUpdater.nano_33_iot.ino.bin -firmware ~/firmwareuploader/firmwares/NINA/1.4.5/NINA_W102.bin -port /dev/ttyACM0 -programmer /usr/bin/bossac`

umbynos, 2021-05-13:
Hi! I just tried on my Pi4  (should be the same architecture)  and it seems to work just fine.
```
uname -a
Linux raspberrypi4 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
```
```
pi@raspberrypi4:~/Documents $ ./FirmwareUploader -flasher firmwares/NINA/FirmwareUpdater.mkrwifi1010.ino.bin -firmware firmwares/NINA/1.4.5/NINA_W102.bin  -address arduino.cc:443 -port /dev/ttyACM0 -programmer ~/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3/bossac 
2021/05/13 12:53:13 Flashing firmware uploader nina
2021/05/13 12:53:13 Entering board into bootloader mode
2021/05/13 12:53:15 Flashing firmwares/NINA/FirmwareUpdater.mkrwifi1010.ino.bin
Atmel SMART device 0x10010005 found
Erase flash
done in 0.636 seconds

Write 13688 bytes to flash (214 pages)
[==============================] 100% (214/214 pages)
done in 0.083 seconds
CPU reset.
2021/05/13 12:53:21 Connecting to programmer
2021/05/13 12:53:21 Opened the serial port with baud rate 115200
2021/05/13 12:53:23 Sync with programmer
2021/05/13 12:53:23 Reading max payload size
2021/05/13 12:53:23 Flashing firmware from 'firmwares/NINA/1.4.5/NINA_W102.bin'
2021/05/13 12:53:23 Erasing 1127424 bytes from address 0x0
Flashing: 99%
2021/05/13 12:54:01 Checking firmware integrity
2021/05/13 12:54:04 Integrity ok
2021/05/13 12:54:04 -----BEGIN CERTIFICATE-----
MIIEZTCCA02gAwIBAgIQQAF1BIMUpMghjISpDBbN3zANBgkqhkiG9w0BAQsFADA/
MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTIwMTAwNzE5MjE0MFoXDTIxMDkyOTE5MjE0MFow
MjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxCzAJBgNVBAMT
AlIzMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAuwIVKMz2oJTTDxLs
jVWSw/iC8ZmmekKIp10mqrUrucVMsa+Oa/l1yKPXD0eUFFU1V4yeqKI5GfWCPEKp
Tm71O8Mu243AsFzzWTjn7c9p8FoLG77AlCQlh/o3cbMT5xys4Zvv2+Q7RVJFlqnB
U840yFLuta7tj95gcOKlVKu2bQ6XpUA0ayvTvGbrZjR8+muLj1cpmfgwF126cm/7
gcWt0oZYPRfH5wm78Sv3htzB2nFd1EbjzK0lwYi8YGd1ZrPxGPeiXOZT/zqItkel
/xMY6pgJdz+dU/nPAeX1pnAXFK9jpP+Zs5Od3FOnBv5IhR2haa4ldbsTzFID9e1R
oYvbFQIDAQABo4IBaDCCAWQwEgYDVR0TAQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8E
BAMCAYYwSwYIKwYBBQUHAQEEPzA9MDsGCCsGAQUFBzAChi9odHRwOi8vYXBwcy5p
ZGVudHJ1c3QuY29tL3Jvb3RzL2RzdHJvb3RjYXgzLnA3YzAfBgNVHSMEGDAWgBTE
p7Gkeyxx+tvhS5B1/8QVYIWJEDBUBgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEE
AYLfEwEBATAwMC4GCCsGAQUFBwIBFiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2Vu
Y3J5cHQub3JnMDwGA1UdHwQ1MDMwMaAvoC2GK2h0dHA6Ly9jcmwuaWRlbnRydXN0
LmNvbS9EU1RST09UQ0FYM0NSTC5jcmwwHQYDVR0OBBYEFBQusxe3WFbLrlAJQOYf
r52LFMLGMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjANBgkqhkiG9w0B
AQsFAAOCAQEA2UzgyfWEiDcx27sT4rP8i2tiEmxYt0l+PAK3qB8oYevO4C5z70kH
ejWEHx2taPDY/laBL21/WKZuNTYQHHPD5b1tXgHXbnL7KqC401dk5VvCadTQsvd8
S8MXjohyc9z9/G2948kLjmE6Flh9dDYrVYA9x2O+hEPGOaEOa1eePynBgPayvUfL
qjBstzLhWVQLGAkXXmNs+5ZnPBxzDJOLxhF2JIbeQAcH5H0tZrUlo5ZYyOqA7s9p
O5b85o3AM/OJ+CktFBQtfvBhcJVd9wvlwPsk+uyOy2HI7mNxKKgsBTt375teA2Tw
UdHkhVNcsAKX1H7GNNLOEADksd86wuoXvg==
-----END CERTIFICATE-----

2021/05/13 12:54:04 Erasing 2048 bytes from address 0x10000
Flashing: 50%
2021/05/13 12:54:04 Operation completed: success! :-)
```

I used bossac installed with [`arduino-cli`](https://github.com/arduino/arduino-cli) with the command  `arduino-cli core install arduino:samd`

umbynos, 2021-09-24:
Closed for inactivity, Feel free to reopen it, if you think the problem is still present with the new version of the tool

closed 2021-09-24 by umbynos