arduino/arduino-fwuploader #43: Error when using the tool for a SAM D21 based Arduino board with WINC1500 WiFi module attached

stevenengland, 2021-06-06:
## Bug Report

### Current behavior

`FirmwareUploader --model winc --flasher FirmwareUpdater.SenseBoxMcuV2.ino.bin --firmware firmware\19.6.1_m2m_aio_3a0.bin --port COM3 --retries 1`

The above command end in this stack trace:
```
2021/06/06 21:00:36 Flashing firmware uploader winc
2021/06/06 21:00:36 Entering board into bootloader mode
2021/06/06 21:00:46 Flashing FirmwareUpdater.SenseBoxMcuV2.ino.bin
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x2b57ed]

goroutine 1 [running]:
github.com/arduino/go-paths-helper.(*Path).String(...)
        /home/runner/go/pkg/mod/github.com/arduino/go-paths-helper@v1.4.0/paths.go:558
github.com/arduino/arduino-cli/executils.NewProcessFromPath(0x0, 0x1145fea0, 0x6, 0x6, 0x2f, 0x0, 0x0)
        /home/runner/go/pkg/mod/github.com/arduino/arduino-cli@v0.0.0-20210422154105-5aa424818026/executils/process.go:54 +0x1d
github.com/arduino/FirmwareUploader/programmers/bossac.(*Bossac).invoke(0x1145ff04, 0x1145fea0, 0x6, 0x6, 0x25, 0x11490000)
        /home/runner/work/FirmwareUploader/FirmwareUploader/programmers/bossac/bossac.go:45 +0x3f
github.com/arduino/FirmwareUploader/programmers/bossac.(*Bossac).Flash(0x1145ff04, 0x11410240, 0x25, 0x0, 0xffffffff, 0x38)
        /home/runner/work/FirmwareUploader/FirmwareUploader/programmers/bossac/bossac.go:37 +0x1c5
github.com/arduino/FirmwareUploader/modules/winc.Run(0x1145ff70, 0x0, 0x0)
        /home/runner/work/FirmwareUploader/FirmwareUploader/modules/winc/main.go:44 +0x564
main.main()
        /home/runner/work/FirmwareUploader/FirmwareUploader/main.go:72 +0x214
```

### Expected behavior

Flasher file is loaded to the Arduino board

### Environment

- Updater version: latest
- OS and platform: Win10

### Additional context

The board I am using is a senseBox MCU v2 (https://en.docs.sensebox.de/hardware/allgemein-sensebox-mcu/) based on SAM D21. In the Arduino IDE I would use arduino:avrispmkii as a programmer but I do not know if this matters here. Does it seem to you like this is a problem with the compiled sketch? Or do I need to follow a specific folder structure (because of the "Path" mentioned in the stack trace).

The bin file is created using an example sketch from the senseBox repo (attached here:
[Winc1500_Updater.zip](https://github.com/arduino/FirmwareUploader/files/6604708/Winc1500_Updater.zip))

stevenengland, 2021-06-07:
Shame on me. I forgot to explicitly set the programmer path via {runtime.tools.bossac}/bossac It might be helpful to mark the command line flags as either optional or mandatory. Without the flag bossac is assumed by the program but then https://github.com/arduino/FirmwareUploader/blob/59fbc283a65303fa0304dcdb85fb15ab503b8653/programmers/bossac/bossac.go line 64 will point to nil.

However, now I have to learn something about why the programmer can't be found. The actual error says
`2021/06/07 15:39:11 Error: exec: "{runtime.tools.bossac}/bossac": file does not exist`
I do not yet know if in your examples this is a placeholder or should "{runtime.tools.bossac}" automatically point to the bossac programmer which is in a dir called something like 
`Arduino15\packages\arduino\tools\bossac\>>1.7.0-arduino3<<\bossac.exe`

KR

umbynos, 2021-06-29:
Hi, unfortunately we removed support for SARA module in #65

closed 2021-06-29 by umbynos

stevenengland, 2021-06-29:
Hi @umbynos thank you for answering. But I am not sure how the mentioned issue is related to my problem. There is no SARA module I need to update. I am using a so called senseBox (https://sensebox.github.io/books-v2/home/en/komponenten/sensebox-mcu.html) which has the ability to plug a WINC1500 WiFi module on it. That's the one I want to update. But I am not quite sure how to do so with the help of the uploader because I do know too little about the programmer to use for example. As far as I understood I only need a working image of the compiled "firmware_uploder.ino" and a working programmer. Isn't that the case?

umbynos, 2021-06-30:
Ups, sorry I misread the issue...

Basically, the update process works in the following way:
1. flash a compiled sketch ([this one](https://github.com/arduino-libraries/WiFi101/blob/master/examples/FirmwareUpdater/FirmwareUpdater.ino)) to the board you want to update
2. then pass the firmware image to the module using serial. (you can find detailed instructions in the source code) 

Unfortunately, we currently do not support other boards. The modules could be different from the ones we use in our boards or could be connected to different pins..

stevenengland, 2021-06-30:
Hi @umbynos , thank you for the clarification. In that case it would be interesting to have a small guidance on how to change the source base to be compatible with my specific board. So that I could download and edit the source for my individual needs. Is there a step by step guide that you follow when adding a new board that you would share with us? So that one like me could follow a recommended way instead of reading all the code and try/guess the correct way of doing that? That would be great! :)

stevenengland, 2021-07-06:
Hi @umbynos , may I open another issue as a feature request either

- for my specific board or better
- for a tutorial that shows how to adapt the code to add new boards (e.g. to use it for PRs or simply to copy, adapt and build locally for ones personal needs)?

I want to build a bigger community of people that measure environmental phenomena but most of them are non technical persons so I want to automate as much as possible including the certificate upload.