arduino/arduino-fwuploader #59: Can't upload firmware to the Arduino Uno WiFi rev.2

LumitoLuma, 2021-06-16:
## Bug Report

### Current behavior
I've tried to upload the NINA firmware (1.4.6) to my Arduino Uno WiFi rev2. When it gets to the `INFO[0008] Opened port /dev/ttyACM0 at 115200` line, the board resets and sends an RX command to the Uno WiFi rev.2 as expected, but then it gets stuck. I have tried it about ten times and I've also disconnected and reconnected the board, but without success.

I've run the following command:
```
./FirmwareUploader firmware flash -v -b arduino:megaavr:uno2018 -a /dev/ttyACM0
```

### Expected behavior
The firmware should upgrade without problems.

### Environment

- Updater version: Git, commit d4c7b34
- OS and platform: Debian Testing x86_64, Linux 5.10.40

### Additional context
I also had a similar problem with previous (stable) releases of the tool: When it tries to connect to the programmer (Uno WiFi rev.2) to start uploading the firmware, the upload fails unless it gets an immediate response from the board.

umbynos, 2021-06-29:
Thanks a lot for the detailed issue. We finally fixed the issue :partying_face:

LumitoLuma, 2021-06-29:
Great. Thanks for your work!

umbynos, 2021-07-02:
FIxed in #74

closed 2021-07-02 by umbynos