arduino/arduino-fwuploader #85: Provide step by step guide to show how to add support for more boards

stevenengland, 2021-07-08:
Hello together, this issue is a continuation due to https://github.com/arduino/arduino-fwuploader/issues/43#issuecomment-874535924

Since this project only supports a selection of boards, it would be a great help for all other board owners if you could provide a step by step guide that shows how to modify existing source to implement support for other boards as well. Either just for forks or local builds or to support PRs/contribution to the project. Does that sound any feasable to you :) ?

Thank you anyway!

umbynos, 2021-07-12:
hello @stevenengland, thanks a lot for the suggestion. We just created a new task (expand our docs with a section called  in our internal issue tracker "dev-docs") inside our internal issue tracker. But we do not have an ETA at the moment.

stevenengland, 2021-07-12:
That's great news! Thank you very much.

davepruitt, 2024-06-12:
Is there any update on this? I frequently make my own custom boards, and I'd like to use the firmware uploader program to update the firmware on the NINA module on my boards. I typically base my boards off of Arduino designs, so they are very similar to Arduino boards (such as the Nano 33 IoT) - but they are not identical.