arduino/arduino-fwuploader #141: panic: runtime error: invalid memory address or nil pointer dereference when listing firmwares

kittaakos, 2022-05-19:
### Describe the problem

This happened in the Arduino IDE. See the original comment [here](https://github.com/arduino/arduino-ide/pull/947#issuecomment-1131351826).

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x14bd30e]

goroutine 1 [running]:
github.com/arduino/arduino-fwuploader/cli/firmware.list(0x0, 0x0)
	/home/runner/work/arduino-fwuploader/arduino-fwuploader/cli/firmware/list.go:66 +0x4e
github.com/arduino/arduino-fwuploader/cli/firmware.newListCommand.func1(0xc000492500, 0xc000158ea0, 0x0, 0x2)
	/home/runner/work/arduino-fwuploader/arduino-fwuploader/cli/firmware/list.go:42 +0x39
github.com/spf13/cobra.(*Command).execute(0xc000492500, 0xc000158e80, 0x2, 0x2, 0xc000492500, 0xc000158e80)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.1.3/command.go:856 +0x2c2
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001eb900, 0x0, 0x0, 0x1501260)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.1.3/command.go:960 +0x375
github.com/spf13/cobra.(*Command).Execute(...)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.1.3/command.go:897
main.main()
	/home/runner/work/arduino-fwuploader/arduino-fwuploader/main.go:35 +0x3c Error: Error executing "/Users/a.kitta/dev/git/arduino-ide/arduino-ide-extension/build/arduino-fwuploader" firmware list --format json: unexpected end of JSON input
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x14bd30e]

goroutine 1 [running]:
github.com/arduino/arduino-fwuploader/cli/firmware.list(0x0, 0x0)
	/home/runner/work/arduino-fwuploader/arduino-fwuploader/cli/firmware/list.go:66 +0x4e
github.com/arduino/arduino-fwuploader/cli/firmware.newListCommand.func1(0xc000492500, 0xc000158ea0, 0x0, 0x2)
	/home/runner/work/arduino-fwuploader/arduino-fwuploader/cli/firmware/list.go:42 +0x39
github.com/spf13/cobra.(*Command).execute(0xc000492500, 0xc000158e80, 0x2, 0x2, 0xc000492500, 0xc000158e80)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.1.3/command.go:856 +0x2c2
github.com/spf13/cobra.(*Command).ExecuteC(0xc0001eb900, 0x0, 0x0, 0x1501260)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.1.3/command.go:960 +0x375
github.com/spf13/cobra.(*Command).Execute(...)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.1.3/command.go:897
main.main()
	/home/runner/work/arduino-fwuploader/arduino-fwuploader/main.go:35 +0x3c
    at ChildProcess.<anonymous> (/Users/a.kitta/dev/git/arduino-ide/arduino-ide-extension/lib/node/exec-util.js:53:31)
    at ChildProcess.emit (node:events:394:28)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)
    at Process.callbackTrampoline (node:internal/async_hooks:130:17)
```

### To reproduce

I do not have a straightforward way to reproduce the defect, but as noted [here](https://github.com/arduino/arduino-ide/pull/947#issuecomment-1096842928), the firmware updater returns with an invalid JSON occasionally.

<details>
  <summary>Click to expand the invalid JSON</summary>
  
```json
[
  {
    "board_name": "Arduino MKR1000",
    "board_fqbn": "arduino:samd:mkr1000",
    "module": "WINC1500",
    "firmware_version": "19.4.4",
    "Latest": false
  },
  {
    "board_name": "Arduino MKR1000",
    "board_fqbn": "arduino:samd:mkr1000",
    "module": "WINC1500",
    "firmware_version": "19.5.2",
    "Latest": false
  },
  {
    "board_name": "Arduino MKR1000",
    "board_fqbn": "arduino:samd:mkr1000",
    "module": "WINC1500",
[... 326 lines cut ...]
arduino-fwuploader Version: 2.0.0 Commit: eb51bf2 Date: 2021-07-28T16:19:28Z

### Operating system

macOS

### Operating system version

12.3.1 (21E258)

### Additional context

I could not reproduce the bug. The latest and 2.0.0 versions worked when I executed the following command in a terminal.

```sh
./arduino-fwuploader firmware list --format json
```

But it happened multiple times in the IDE. I do not know where the null pointer exception comes from.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-fwuploader/issues?q=)
- [X] I verified the problem still occurs when using the latest version
- [X] My report contains all necessary details

umbynos, 2022-05-19:
Hi @kittaakos could you please try if the problem is still present in the latest release ([2.2.0](https://github.com/arduino/arduino-fwuploader/releases/tag/2.2.0) ATM). I'm asking because this probably has been fixed in #128

kittaakos, 2022-05-19:
Thank you for the quick reaction ❤️ 

> could you please try if the problem is still present in the latest release ([2.2.0](https://github.com/arduino/arduino-fwuploader/releases/tag/2.2.0) ATM)?

I tried it with both 2.0.0 and 2.2.0:

> I could not reproduce the bug. The latest and 2.0.0 versions worked when I executed the following command in a terminal.
> 
> ```shell
> ./arduino-fwuploader firmware list --format json
> ```
> 
> But it happened multiple times in the IDE. I do not know where the null pointer exception comes from.

I promise to close this issue once the IDE2 uses the latest (2.2.0) version, and I do not see the panic anymore. Unfortunately, I have no idea how to reproduce this. It happened only twice so far, from the IDE2 from debug mode.

----
Feel free to keep the _waiting for information_ label.

per1234, 2022-05-19:
I have also observed this panic periodically when starting the IDE.

I was never able to find how to reproduce it consistently though, even with manually corrupting the index, so I couldn't investigate.

kittaakos, 2022-06-07:
I am closing it. I have not seen the same error with the `2.2.0` version in IDE2.

closed 2022-06-07 by kittaakos